nanoframework/nf-interpreter #115: Add documentation about cmake-variants.TEMPLATE.json

josesimoes, 2017-02-06:
- Add cmake-tools-cmake-variants.md to docs\cmake
- Explain the purpose of the file, the various entries and how a developer should tweak it.
- Add link to it on the items list at the upper level main document.

MikroBusNet, 2017-04-07:
Is the following draft what you expect ?
[cmake-tools-cmake-variants.zip](https://github.com/nanoframework/nf-interpreter/files/906688/cmake-tools-cmake-variants.zip)

josesimoes, 2017-04-17:
> Is the following draft what you expect ?

yes, that is quite all right! I would just add (at the end, right after the example configuration) a list explaining what each entry there is meant for.

(in case you are not sure on any of those, just leave it empty and it will be filled in latter)

closed 2017-04-17 by MikroBusNet