nanoframework/nf-interpreter #121: Use of CORTEX_USE_FPU TRUE hangs MCU

MikroBusNet, 2017-02-15:
In chconf.h, setting CORTEX_USE_FPU to TRUE compiles fine but hangs the board when flashed on a STM32F427 MCU using ChibiOS.

Please see [this thread](https://nanoframework.slack.com/archives/chibios/p1487075678000106) for more detailed informations if needed.

cw2, 2017-02-28:
The problem here is that nF CMake build system does not pass `CORTEX_USE_FPU` symbol properly to ChibiOS build. After adding `-DCORTEX_USE_FPU=TRUE` to compiler and assembler command line options, the FPU works just fine.

CMake/toolchain.ChibiOS.GCC.cmake
```
set(CMAKE_C_FLAGS_DEBUG "-DCORTEX_USE_FPU=TRUE ... )
set(CMAKE_CXX_FLAGS_DEBUG "-DCORTEX_USE_FPU=TRUE ... )
set(CMAKE_ASM_FLAGS_DEBUG "-DCORTEX_USE_FPU=TRUE ... )
```
Note: The most important is the assembler, because the FPU initialization happens in `ChibiOS\os\common\ports\ARMCMx\compilers\GCC\crt0_v7m.s` - without it, any use of FPU (VFP) instruction results in exception.

piwi1263, 2017-03-01:
@cw2 or any who can answer this.

How should the -mfloat-abi=...... be set ? 
If they are to be set to -mfloat-abi=hard must it be for all flags, C, CXX, ASM, LINKER, etc. in <repo-root>\CMake\Modules\CHIBOS_STM32F4xx_GCC_options.cmake ?

cw2, 2017-03-01:
If I understand it correctly, it is done via CMake option:
```
cmake ... -DUSE_FPU=TRUE ...
```
Generally, you should not need to modify any internal CMake files - ideally, all configurable settings should be specified via command line options.

cw2, 2017-03-01:
@piwi1263  I am going to merge this PR momentarily, so you can use it as described above (and also in the updated documentation) after that.

PR #140 (2017-03-01): Add support to allow use of hardware FPU
- add new parameter to main CMake: USE_FPU
- update CMake GCC option files for F4 and F series
- update CMake for reference boards where applicable
- update documentation accordingly
- update (and clean up) cmake-variants template file

Fixes #121
commit 6ad95137 (2017-03-01): Add support to allow use of hardware FPU (#140)

Add support to allow use of hardware FPU

- add new parameter to main CMake: USE_FPU
- update CMake GCC option files for F4 and F series
- update CMake for reference boards where applicable
- update documentation accordingly
- update (and clean up) cmake-variants template file

Fix #121
closed 2017-03-01 by cw2