nanoframework/nf-interpreter #124: I2C support fails to compile on F42x MCU

MikroBusNet, 2017-02-16:
As soon as you add I2C support for F42x boards (F427 & F429 so far), you get compilation errors about many missing declarations (e.g. I2C_CR2_ADD10) : 

> C:\dev\nf-interpreter\build\ChibiOS_Source\os\hal\ports\STM32\LLD\I2Cv2\i2c_lld.c: In function 'i2c_lld_set_address':
C:\dev\nf-interpreter\build\ChibiOS_Source\os\hal\ports\STM32\LLD\I2Cv2\i2c_lld.c:137:28: error: 'I2C_CR2_ADD10' undeclared (first use in this function)
   if ((i2cp->config->cr2 & I2C_CR2_ADD10) == 0U)

Such declarations are available for many other MCUs but not F42x ones.

MikroBusNet, 2017-02-19:
Issue sorted out :)

In fact, F42x needs I2CV1, not I2CV2 !
Once the file CHIBIOS_STM32F4xx_sources.cmake should reference I2CV1 has been edited accordingly (2 lines), everything went fine. At least there were no compilation error.

That said, some other STM32F4 board may need I2CV2. This has to be checked, I think.

josesimoes, 2017-02-20:
Nice find! Now we know what needs to be done to have it working.
The CMake for F4 have to include one module or the other according yo the device. Adding a couple of IFs there should take care of this.

MikroBusNet, 2017-02-26:
A fix for STM32F427/F429 is present in this [PR](https://github.com/nanoframework/nf-interpreter/pull/132).

josesimoes, 2017-03-16:
@MikroBusNet please confirm that this was indeed fixed in #168. 
If yes, please update the labels and close this one.

MikroBusNet, 2017-03-25:
Woops !
Indeed, it is fixed in PR#168. A better fix will come in the near future, but the one coded in the PR is working fine, no worry.

closed 2017-03-25 by josesimoes