nanoframework/nf-interpreter #131: Using floats in ChibiOS chprintf hangs an STM32F4xx board

piwi1263, 2017-02-25:
When using the chprintf function to print a float by using %f effectively freezes the MCU. In order to replicate the issue you have to set CHPRINTF_USE_FLOAT in chprintf.h to TRUE.

[image]

cw2, 2017-02-26:
If I understand it correctly, the build is configured to use newlib-nano - and it by default does not have floating point support enabled, it has to be done via `-u _printf_float` option.

piwi1263, 2017-02-26:
@cw2, thanks, I\ll give it a shot ....

piwi1263, 2017-02-26:
In the CMAKE_C_FLAGS a -specs=nosys.specs needs to be included. Then, at least the compiler and the linker are happy, otherwise you get a linker error that _sbrk is not defined. Tip for @Becafuel

josesimoes, 2017-02-26:
@piwi1263 that bring a lot more stuff other than just printf... Compare the size of the image built with nonsys against a build with nanolib.
My point being do we need floating point output that badly so we need to quit on nanolib?
There are ways of making that available sticking with nanolib, which I believe we should...

piwi1263, 2017-02-26:
Even after all set to hard FPU use and using the above the system flags for the C compiler settings, it does manage to go to the first chprintf statement. It arrives and stops there. Running into the same exception as reported above in the screen shot.

josesimoes, 2017-02-26:
Hardware FPU is not working right now. That's been reported at #121.

piwi1263, 2017-02-26:
After some research it basically starts an exception executing the following statement in chcoreasm_v7m.s

Executing (F10 or F11) will effectively stop the MCU from responding while (true) ...

[image]

josesimoes, 2017-02-26:
Yes. That's what's reported on #121 and possibly the root cause for this. Not so much the lack of support for it _per se_ in the runtime.

MikroBusNet, 2017-02-26:
> My point being do we need floating point output that badly so we need to quit on nanolib? There are ways of making that available sticking with nanolib, which I believe we should...

I do not know if we really need to print/display floats or double from inside the native parts. I will personnaly ignore them as much as I can as I do not want to loose time on this at the moment. There are other issues more important than this one, to me.

josesimoes, 2017-03-01:
@piwi1263 when #140 is merged please check if the CPU hang-up reported above gets corrected or not.

cw2, 2017-03-01:
PR #140 merged.

piwi1263, 2017-03-01:
Although I change a bit more since my board has the st-link part cut off and had to output to usart6 I can confirm it does work, see picture below. We have a PI printed. @christoph I reckon this solves both our issues.

Because of using chprintf I had to include the chibios-repo-root\os\hal\lib\streams to the include paths.
Set CHPRINTF_USE_FLOAT to TRUE in chprintf.h in the same directory.

Added three lines to repo-root\..\FindCHIBIOS.Make:
1. list(APPEND CHIBIOS_INCLUDE_DIRS ${PROJECT_BINARY_DIR}/ChibiOS_Source/os/hal/lib/streams) (~ Line 99)
2. chprintf.c @ ~ Line164
3. ${PROJECT_BINARY_DIR}/ChibiOS_Source/os/hal/lib/streams (~Line 185)

Change board.h in the chibios\os\hal\boards\your-board to set LSE to 32768 and HSE to 8000000
Change the compiler flag as described in #121 
If you run into an _brk... not defined add -specs=nosys.specs in <root>\CMake\Modules\CHIBIOS_STM32F4xx_GCC_options.cmake to the C flags
Added in the same file and flags options -u _printf_float

[image]

josesimoes, 2017-03-02:
Closing this as it seems to be solved with proper FPU configuration.

closed 2017-03-02 by josesimoes