nanoframework/nf-interpreter #136: ESP32 Support

cw2, 2017-02-27:
I am happy to announce that ESP32 support is now under active development.

The implementation is based on the official [Espressif IoT Development Framework](https://github.com/espressif/esp-idf) (ESP-IDF), initially for a generic [ESP-WROOM-32](https://espressif.com/en/producttype/esp-wroom-32) module.

The following is a rough roadmap:

- [x] Initial experiments (toolchain setup, build environment, deployment, debugging, diagnostics)
- [ ] nanoBooter port
  - [ ] :question: implement as a part of 2nd stage bootloader or a separate [3rd stage] ?
- [ ] :running_man: nanoCLR port

> Note: The nanoCLR is being done now (before bootloader) to synchronize with ongoing ChibiOS HAL work.

ESP-IDF has many features that make it very good fit for nanoFramework (the list is not complete, and in no particular order of importance):
- GCC-based toolchain,
- Highly componentized and customizable build system,
- Bootloader is a separate application,
- Customizable flash memory partition table,
- Over-the-air updates

josesimoes, 2017-02-27:
👏  👏   👏

piwi1263, 2017-02-27:
👏👏👏

If you need a hand in testing out or if you need a guinnee pig just let me know ....

josesimoes, 2017-05-11:
Could you please confirm if this actually _in progress_ or not?

closed 2017-09-22 by MikroBusNet