nanoframework/nf-interpreter #141: USB device doesn't show in Device Manager for F411RE and F427

josesimoes, 2017-03-01:
Despite the ChibiOS driver configuration (following reference for Discovery4)  and hardware configuration (using PA11 and PA12 pins) these boards don't show in Windows Device manager. It doesn't seem to be a fault in the configuration because the USB bus seems to don't even try to enumerate it.

MikroBusNet, 2017-03-03:
Nothing very constructive in my comment, but to me this is something that will be mandatory to address at some time. 
Boards like Netduino will show the same behaviour, for example. And "omitting" such users base will be problematic, I think.
Also, I think that GHI's little boards like Lemur, Panda and Brainpad will be in the same bag regarding USB.

MikroBusNet, 2017-03-04:
A fix is waiting for review in PR #132

josesimoes, 2017-03-05:
@MikroBusNet nice finding with this! 👏 

For the record: the root cause for this is that the default configuration for the OTG1 module is such that the USB enumeration is triggered only after a valid electrical connection occurs. This is detected by sensing the Vbus of the USB connector. Meaning that the device needs to be connected to an USB host, so the 5V are present on the Vbus signal triggering the USB enumeration workflow. 
This requires that besides OTG_FS_DM and OTG_FS_DP also the OTG_FS_VBUS signal is routed and properly configured.

The OTG1 module can be forced to be an USB "peripheral only" device not requiring the Vbus presence/sensing. This happens by setting a bit in the Global USB configuration register (FDMOD in
OTG_FS_GUSBCFG). The ChibiOS driver handles this through a define. Setting `#define BOARD_OTG_NOVBUSSENS ` preferably in the board.h file takes care of this.

closed 2017-03-05 by josesimoes