nanoframework/nf-interpreter #143: Add support for streams and ChibiOS printf

josesimoes, 2017-03-02:
See last comment of @piwi1263 on #131.

- [ ] add source code for streams

- [ ] add include directory(es) supporting streams

- [ ] investigate use of newlib-nano instead of nosys.specs and the need for -u _printf_float

piwi1263, 2017-03-02:
I'll kill the additional options mentioned in point 3 and see what is really minimal needed. But as you mentioned already if there is no better way to achieve this. The ending address of nano ooter comes dangerously close to the nanoCLR start address.

josesimoes, 2017-03-02:
> I'll kill the additional options mentioned in point 3 and see what is really minimal needed. 

Nice!

> The ending address of nano ooter comes dangerously close to the nanoCLR start address.

No problem. We can change that address at any time. 
Because they all depend on the others and not being able to come up with a smart way of adjusting all those addresses on an automated fashion we have to that manually. At least for now.
I don't think this is worth more work than that, as the nanoBooter is something that is supposed to be quite stable and not requiring release after release.

cw2, 2017-03-02:
@piwi1263 What do you need to do in nanoBooter that it requires [full] newlib? The only thing I can think of is use of long-double and 64-bit integers, which is not supported in newlib-nano (and require custom build; this will bubble up later). Even support for standard floating point is included, but only not linked by default - it must be 'enabled' explicitly by that special symbol.

piwi1263, 2017-03-03:
@cw2 My needs, there is nothing of any need since nanoBooter will just do fine without this float printing. This is question on one of slack channels why %f prints do not work or hold the system. So whomever needs this, it is a way to do it but at a little price.

So as a final test I did a test not using chprintf but the standard sprintf() and put that resulting buffer to serial using streamWrite() it is not as simple as chprintf but does the job. Plus, I haven't tested that, but I reckon that sprintf is supporting more options (%whatever) than chprintf.

chprintf requires more changes to the source files and more includes in the main.c or where ever you need to use that. They pretty much have the same memory footprint. For both you need to enabled FPU of course and the -u _printf_float -specs=nosys.specs in the c compiler options in the CHIBIOS_<MCU>_GCC_options. And of course the -DCORTEX_USE_FPU=TRUE option added to the C, CXX, ASM flags in toolchain.ChibiOS.GCC.cmake

In any case if someone has questions concerning undefined _sbrk errors or float printing support issues for what reason whatsoever they can be forwarded to #143 or #131 or #121 for further reading

Maybe obvious for others but just to complete my findings I used following code snippet from @Becafuel to test successfully the sprintf way of things. It requires an _extern SerialDriver SD2_ declare, I used usbcfg.h and in my case not SD2 but SD6

[image]

josesimoes, 2017-03-03:
> And of course the -DCORTEX_USE_FPU=TRUE option added to the C, CXX, ASM flags in toolchain.ChibiOS.GCC.cmake

After #140 you don't need to add anything to the CMake toolchain file. All the FPU handling/config is taken care of by setting the CMake parameter -DUSE_FPU=TRUE.

piwi1263, 2017-03-03:
Oh, that's convenient. Mind telling me where I need to apply that ? And do I still need to set other FPU options like in the chconf.h in nanoBooter, nanoCLR of the target I use ?

piwi1263, 2017-03-03:
Found it, very nice solution for this. 👍  I'd better look before I ask ... 🙈 

If someone thinks to use this float print the '-u _printf_float' still needs to be added

closed 2017-09-15 by josesimoes