nanoframework/nf-interpreter #228: Add command to WireProtocol to report detailed flash map to host

josesimoes, 2017-03-23:
Need to send the flash map with block detals to allow the host to manage the flash space.

MikroBusNet, 2017-03-26:
Should the exact memory consumption be detailed in this map or only allocated blocks ?

e.g. 
```
0x08000000 - 0x08007FFF : nanoBooter
0x08008000 - 0x08CFFFFF : nanoCLR
```
or
```
0x08000000 - 0x080049A0 : nanoBooter
0x08008000 - 0x0805920E : nanoCLR
```

Also, should it display only the summary of used memory like just above or should it detail used blocks like below ?

```
0x08000000 - 0x08003FFF : nanoBooter  (16 K)
0x08004000 - 0x080062A7 : nanoBooter (16 K)
0x08008000 - 0x0800BFFF : nanoCLR (16 K)
0x0800C000 - 0x080FFFFF : nanoCLR (16 K)
0x08010000 - 0x0804FFFF : nanoCLR (128 K)
0x08050000 - 0x0808FFFF : nanoCLR (128 K)
```

josesimoes, 2017-03-26:
Great question!
Following the RFC0001, and to serve the requirement of the host being able to manage the flash space,   this should report the *regions* (CLR and deployment) _and_ details about each region, in particular, how many blocks and their size. Addressing can probably be inferred from the above, so there is no need to send it.

All this seems to be possible by using the definition of BlockStorage. As we don't have (yet) a better way for managing that it will have to work for this.
What is clearly missing here is a simple tool to "auto" generate the block storage mapping. The current "manual" way of doing it requires some knowledge on how the structure works *and* also understanding the memory  map of the device *and* how all this stuff of CLR and deployment works.
For me that can be a show stoper for new comers and even for developers lacking the understanding of the tech details.
This might be a nice addition to ANT... kind of a simplistic visual tool that spits the code for the block storage struct, ready for copy/paste on to a file.

piwi1263, 2017-03-26:
Would be a nice job for having the details to be retrieved via a webjob service (microservice) with input of the mcu and the mapping structure coming back. Bit by bit the back-end db (whatever you choose to use) gets filled up by the individual mcus as the come avail or entered in the db by those who know what they are doing.

PR #255 (2017-03-28): Work on Debugger
- plug Debugger with WireProtocol in plain C
- fix #226
- completly replaced WireProtocol C++ with C version
- replace private_malloc with platform_malloc
- rework Monitor_FlashSectorMap to report full details on BlockStorage
- fix #228
- further improvements in BlockStorage
- add stubs for COM
- add interface from BlockStorage interface to STM32 Flash driver
- rework nanoCLR VS Solution to make it compatible with latest changes
- revert several types to well know types
- add weak declarations to COM stubs in Win32
- correct linker scripts to have addresses aligned with sector [... cut]
commit 94212714 (2017-03-28): Work on Debugger (#255)

* Work on Debugger

- plug Debugger with WireProtocol in plain C
- fix #226
- completly replaced WireProtocol C++ with C version
- replace private_malloc with platform_malloc
- rework Monitor_FlashSectorMap to report full details on BlockStorage
- fixes #228
- further improvements in BlockStorage
- add stubs for COM
- add interface from BlockStorage interface to STM32 Flash driver
- rework nanoCLR VS Solution to make it compatible with latest changes
- revert several types to well know types
- add weak declarations to COM stubs in Win32
- correct linker scripts to have [... cut]
closed 2017-03-28 by josesimoes