nanoframework/nf-interpreter #237: Update documentation on how to add new boards for ChibiOS

josesimoes, 2017-03-24:
This is in docs/CMake/chibios-build.md.

When adding support for a new board in ChibiOS a developer can choose to add the _board.c_ and _board.h_ files either on the overlay folder or right on the target board folder.
The choice between the two options is simple: if you think it's going to be reused on many target boards, add it to the overlay; if it's a one shot only, then add those right on the target board folder.

MikroBusNet, 2017-04-07:
FWIW, I think that each board is "unique" because of their different pinouts. Indeed, the MCU may be the same across different boards but pin mappings are almost always different. e.g. Nucleo F746 and F746 Discovery are completely different boards.

So I would not choose and put all the *board.c/board.h* files in one folder, that would then be the __target board__ folder.
This is only my opinion, however, I may be missing something else.

josesimoes, 2017-04-17:
> FWIW, I think that each board is "unique" because of their different pinouts

Absolutely! 

> So I would not choose and put all the board.c/board.h files in one folder

We are in agreement here. Right now there are 3 possible locations for the board.c/board.h files:

1. The source folder of ChibiOS
2. The nanoFramework community overlay at \targets\CMSIS-OS\ChibiOS\nf-overlay\os\hal\boards\NNNN (pretty much equivalent to 1.)
3. Right on the target board folder at \targets\CMSIS-OS\ChibiOS\MBN_QUAIL (for example)

The last one is meant to ease the build process, not only for the community supplied boards but for any target board that is rather unique. Usually if one is using a CPU that will be reused over and over the place for those is 2. (or 1. if contributed back to ChibiOS source).

closed 2017-09-22 by MikroBusNet