nanoframework/nf-interpreter #254: Hard coding of flash sectors in flash_lld_erase function

piwi1263, 2017-03-27:
The erase function uses hard coded FLASH_SECTOR_11 which will not be avail for MCUs having lesser flash sectors and the compile stops with an undeclared error.

Below find a screenshot of a compile for the build based upon a board with MCU STM32F411RE. The MCU contains only 7 flash sectors so FLASH_SECTOR_11 will not be declared.
[image]

I suggest to use FLASH_SECTOR_MAX in the comparison statement and defined for all variations of the maximal flash sectors for the various MCUs in the accompanying header file.

josesimoes, 2017-03-27:
Please note that there are 2 versions of the low level drivers. Those are supposed to address most, if not all, the possible variations. The diference in the available flash sizes is taken into account there. Please check if the 411 isn't already covered, It should be...

piwi1263, 2017-03-30:
No it is not since it has only 7 flash sectors so flash sector 11 is not defined and runs into a compile error -> ergo no build. If this is to be generic code to cover all the MCUs it should not be declared in some desolate header but addressed in the flash_lld logic. How ? Is per discussion.

piwi1263, 2017-03-30:
Don't no if this code piece covering the FLASH_SECTOR_11 is because of the latest sector being used for a single or few MCUs or across the board for all MCUs that if used sector is higher than last sector defined it has to do the if block. Maybe the if block needs to be #ifdefined FLASH_SECTOR_11 compiler directive.
[image]

Or if the if block has to be executed for the last sector defined for each and every MCU then I'd suggest to go for a FLASH_SECTOR_LAST define per MCU define block in the header file and use the FLASH_SECTOR_LAST in the if block.
[image]

closed 2017-09-15 by josesimoes