nanoframework/nf-interpreter #413: Port and build for STM32F411RE

josesimoes, 2017-07-29:
@valoni as I didn't want to hijack the .NETMF issue, so I'm opening this here were it belongs.

Regarding your questions on NETMF/netmf-interpreter#538:

- indeed there is a community board for the STM32F411RET6, @piwi1263 is the author
- you can fin the build instructions along with other documentation in the docs folder [here](https://github.com/nanoframework/nf-interpreter/tree/master/docs)
(you'll find that you don't nee much 😉 )
- as the documentation is a bit scarce you'll probably need some guidance suggest you join our Slack community by filling in the form [here](https://nanoframework.wordpress.com/slack-invite-form)
- you'll the VS 2017 extension too that you can install directly in VS by searching "nanoFramework" in the Tools and Extensions Window.

Trust this can, at least, get you started. Anyways you'll get all the help that you ask for here on in the Slack community.

josesimoes, 2017-08-09:
@valoni how's your progress on this?

valoni, 2017-08-09:
a little busy with drivers for tinyclr for rfid rdm6300 and esp8266 

and still for me strange to use vs code at least by cmd mode and lot of manually work for tweaking

valoni, 2017-08-09:
but i belive until Sunday i will be able to setup properly my pc to be able to compile at least.

josesimoes, 2017-08-09:
Ok. But know that you only need VS Code if you plan to work on the code. In order to build an image you only need GCC toolchain and CMake.

valoni, 2017-08-10:
Configured everything by 
https://github.com/techcap/nf-interpreter/blob/master/docs/quick-start-guide.md

info and setuped everything as is on the file but still nothing :( happened or missing any hints

closed 2017-09-05 by josesimoes