nanoframework/nf-interpreter #414: F411re compiling issue

nbsn2, 2017-08-07:
```
C:\nf-interpreter\targets\CMSIS-OS\ChibiOS\nf-overlay\os\hal\ports\STM32\LLD\FLASHv2\flash_lld.c: In function 'flash_lld_getSector':
C:\nf-interpreter\targets\CMSIS-OS\ChibiOS\nf-overlay\os\hal\ports\STM32\LLD\FLASHv2\flash_lld.c:276:22: error: 'ADDR_FLASH_SECTOR_8' undeclared (first use in this function)
   else if((address < ADDR_FLASH_SECTOR_8) && (address >= ADDR_FLASH_SECTOR_7))
                      ^~~~~~~~~~~~~~~~~~~
```
there is no in ADDR_FLASH_SECTOR_8 defined in flash_lld.h

```
#if defined(STM32F401xE) || defined(STM32F411xE) || defined(STM32F446xx)

#define FLASH_SECTOR_0     ((uint32_t)0U) /*!< Sector Number 0   */
#define FLASH_SECTOR_1     ((uint32_t)1U) /*!< Sector Number 1   */
#define FLASH_SECTOR_2     ((uint32_t)2U) /*!< Sector Number 2   */
#define FLASH_SECTOR_3     ((uint32_t)3U) /*!< Sector Number 3   */
#define FLASH_SECTOR_4     ((uint32_t)4U) /*!< Sector Number 4   */
#define FLASH_SECTOR_5     ((uint32_t)5U) /*!< Sector Number 5   */
#define FLASH_SECTOR_6     ((uint32_t)6U) /*!< Sector Number 6   */
#define FLASH_SECTOR_7     ((uint32_t)7U) /*!< Sector Number 7   */

#define ADDR_FLASH_SECTOR_0     ((uint32_t)0x08000000) /* Base @ of Sector 0, 16 Kbyte */
#define ADDR_FLASH_SECTOR_1     ((uint32_t)0x08004000) /* Base @ of Sector 1, 16 Kbyte */
#define ADDR_FLASH_SECTOR_2     ((uint32_t)0x08008000) /* Base @ of Sector 2, 16 Kbyte */
#define ADDR_FLASH_SECTOR_3     ((uint32_t)0x0800C000) /* Base @ of Sector 3, 16 Kbyte */
#define ADDR_FLASH_SECTOR_4     ((uint32_t)0x08010000) /* Base @ of Sector 4, 64 Kbyte */
#define ADDR_FLASH_SECTOR_5     ((uint32_t)0x08020000) /* Base @ of Sector 5, 128 Kbyte */
#define ADDR_FLASH_SECTOR_6     ((uint32_t)0x08040000) /* Base @ of Sector 6, 128 Kbyte */
#define ADDR_FLASH_SECTOR_7     ((uint32_t)0x08060000) /* Base @ of Sector 7, 128 Kbyte */

#endif /* STM32F401xE || STM32F411xE || STM32F446xx */
```
so I added 
#define ADDR_FLASH_SECTOR_8     ((uint32_t)0x08080000) /* Base @ of Sector 8, 128 Kbyte */
into .h file.

Current compiling error log:
```
c:/gcc-arm-none-eabi-6-2017-q1-update-win32/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: nanoBooter.elf section `.rodata' will not fit in region `flash'
c:/gcc-arm-none-eabi-6-2017-q1-update-win32/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: region `flash' overflowed by 1468 bytes
collect2.exe: error: ld returned 1 exit status
```
It's a long last issue.

josesimoes, 2017-08-07:
@nbsn2 thanks for spotting that one. The flash driver was revised a few weeks ago and this one slipped. 

The F411 doesn't have sector 8 (and 0x08080000 is an invalid address) so adding that solved the compiling issue but is not technically correct. Need to rework the #if #def blocks there.

As for the  **region `flash' overflowed ** error that is because the code you have there won't fit in the available flash size. Not sure what is the compile option you are using but suggest that you try to switch to "RELEASE" instead of "DEBUG". This way the linker will improve the code and you'll probably will make it.

Know that one of our goals is to improve the code and make it more 'modular' so the target devices with smaller memories are able to run nanoFramework too.

josesimoes, 2017-08-07:
@nbsn2 you've probably noticed that but know that the target you are working is on the community targets repo [here](https://github.com/nanoframework/nf-Community-Targets/tree/master/CMSIS-OS/ChibiOS/ST_NUCLEO_F411RE_EX).

josesimoes, 2017-08-07:
@nbsn2 the sector 8 error should now be fixed. Please confirm.

nbsn2, 2017-08-07:
Check ok on  > sector 8 issue
But still can't compile through.
Although the Doc mentioned 
`CMAKE_BUILD_TYPE: build type (Debug, Release, etc). The default is Release.
`
It seams should be explicitly set.

So the current cmake set:

`cmake -DTOOLCHAIN_PREFIX="C:/GNU_Tools_ARM_Embedded" -DCMAKE_BUILD_TYPE=RELEASE  -DCHIBIOS_SOURCE=C:/ChibiOS-master -DCHIBIOS_BOARD=ST_NUCLEO_F411RE_EX -DTARGET_SERIES=STM32F4xx -DUSE_FPU=FALSE -DNF_FEATURE_DEBUGGER=FALSE  -DAPI_Windows.Devices.Gpio=ON -DNF_FEATURE_RTC=OFF -G "NMake Makefiles" ../
`
nanoBooter.elf is built out , but not smoothly.

```
[ 30%] Linking C executable nanoBooter.elf
   text    data     bss     dec     hex filename
  15084     508  131024  146616   23cb8 nanoBooter.elf
Generate nanoBooter HEX and BIN files for deployment, LST file for debug
C:\GNU_Tools_ARM_Embedded\bin\arm-none-eabi-objcopy.EXE: C:/nf-interpreter/build/nanoBooter.elf: section `.mstack' can't be allocated in segment 2
LOAD: .bss .mstack .pstack .heap
C:\GNU_Tools_ARM_Embedded\bin\arm-none-eabi-objcopy.EXE: C:/nf-interpreter/build/nanoBooter.elf: section `.pstack' can't be allocated in segment 2
LOAD: .bss .mstack .pstack .heap
[ 30%] Built target nanoBooter.elf 

```

nanoCLR.elf build failed.

```
[100%] Linking CXX executable nanoCLR.elf
C:\Users\c\AppData\Local\Temp\ccqGlREQ.ltrans27.ltrans.o: In function `ReceiverThread':
<artificial>:(.text.ReceiverThread+0x38): undefined reference to `WP_Message_Initialize'
<artificial>:(.text.ReceiverThread+0x3c): undefined reference to `WP_Message_PrepareReception'
<artificial>:(.text.ReceiverThread+0x40): undefined reference to `WP_Message_Process'
collect2.exe: error: ld returned 1 exit status
NMAKE : fatal error U1077: “C:\GNU_Tools_ARM_Embedded\bin\arm-none-eabi-g++.exe”: Return code “0x1”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.10.25017\bin\HostX86\x86\nmake.exe"”: Return code “0x2”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.10.25017\bin\HostX86\x86\nmake.exe"”:Return code “0x2”
Stop.
```

josesimoes, 2017-08-07:
The warnings about _.mstack' can't be allocated --_ can safely be ignored. Don't worry about those. And you have a good nanoBooter image now! 👏 

The errors on nanoCLR are there because you are disabling the debugger feature with -DNF_FEATURE_DEBUGGER=FALSE.
At this this time there is no "minimal" version of the debugger so you really have to enable it.

nbsn2, 2017-08-07:
tried 
```
-DNF_FEATURE_DEBUGGER=TRUE \
-DAPI_Windows.Devices.Gpio=ON \
-DNF_FEATURE_RTC=ON
```

Now everything is ok.
Thanks a lot.

PR #415 (2017-08-07): Correct issue with flash driver for STM32F4XX devices
- fixes #414
commit aed66d91 (2017-08-07): Correct issue with flash driver for STM32F4XX devices (#415)

- fixes #414
closed 2017-08-07 by nbsn2