nanoframework/nf-interpreter #420: Code is redudant and useless

MikroBusNet, 2017-08-10:
https://github.com/nanoframework/nf-interpreter/blob/3d39a3291f12d69e9bf105002a24fe003a16d994/targets/CMSIS-OS/ChibiOS/nanoCLR/Windows.Devices.Gpio/windows_devices_gpio_native_Windows_Devices_Gpio_GpioPin.cpp#L156

The check for pin disposed has already been done in the managed code. So the native method will not ever be called.

josesimoes, 2017-08-10:
Checking the class dispose flag object in the managed code is **waay** less efficient than doing it on the native end. That's why I've moved it there.

MikroBusNet, 2017-08-10:
I do not know which one is more efficient but the native one will not be called, here.

josesimoes, 2017-08-10:
You are right! Guess that the check for disposed on the native side is kind of a paranoid approach to double check the dispose status. 

It should be removed. 👍

josesimoes, 2017-08-21:
The check for disposed object on native should be left there. In case the function is called from the execution engine or from an event handler.

closed 2017-08-21 by josesimoes