nanoframework/nf-interpreter #427: Bad SPI behaviour after PR#425

MikroBusNet, 2017-08-12:
PR#425 moved the CLRStartup into its own OS thread. However, this cause DMA failures when SpiSend() is called from within the Devices.Spi assembly (native side).

Reverting back to the previous code (CLRStartup called from the main thread) did fix the issue.

josesimoes, 2017-08-19:
@MikroBusNet I believe this working fine now. Can this issue be closed?

MikroBusNet, 2017-08-21:
I have found that the osKernelStart() method needs some time to "stabilize" before calling any subsequent code.
The SPI code, for example, is either failing or giving unexpected results if it is initialized right after the osKernelStart() method. If I add a osDelay(50) after osKernelStart(), then everything goes fine.

Perhaps the 50ms value could be lowered, however. This was just a quick and dirty test to validate the fix.

josesimoes, 2017-08-21:
That call to osDelay() makes the OS threads run. The SPI init is probably making changes that need to wait for something to be settled in the very first run of the thread pool management. 
Because any SPI stuff is called somewhere else in the code and not on main() I'm closing this as it shouldn't be of concern.

closed 2017-08-21 by josesimoes

MikroBusNet, 2017-08-21:
If SPI stuff is called from inside the Devices.Gpio API (native side), then the bad behaviour comes back again.
If I use the old way of starting the CLR, then everything is fine in Devices.Gpio.
So I would not close this issue...