nanoframework/nf-interpreter #496: Debug and Release - sometime corrupt firmware with deploy from VS

valoni, 2017-09-11:
Tested and done with community boards Nucleo STM32F411RET6 

if i compile with RELEASE mode it seems not stable when i deploy after some time applications 
(tested with simple blink code) it corrupt firmware so i need to reflash again it.

if i compile with DEBUG MODE it seems more stable and i can deploy well but is not tested heavy to be sure to not corrupt

if tried to deploy with other (earlier dll) it corrupt firmware at the moment it was deployed

networkfusion, 2017-09-11:
@valoni please add the board(s) that you are trying, also have you tried MinSizeRel || RelWithDebugInfo. Also, attaching your cmake-variants.json will be of help.

valoni, 2017-09-11:
{
  "buildType": {
    "default$": "debug",
    "description$": "the build type to use",
    "debug": {
      "oneWordSummary$": "Debug",
      "description$": "Emit debug information without performing optimizations",
      "buildType": "Debug"
    },
    "release": {
      "oneWordSummary$": "Release",
      "description$": "Enable optimizations, omit debug info",
      "buildType": "Release"
    },
    "minsize": {
      "oneWordSummary$": "MinSizeRel",
      "description$": "Optimize for smallest binary size",
      "buildType": "MinSizeRel"
    },
    "reldeb": {
      "oneWordSummary$": "RelWithDebInfo",
      "description$": "Perform optimizations AND include debugging information",
      "buildType": "RelWithDebInfo"
    }
  },
  "linkage": {
    "ST_STM32F4_DISCOVERY": {
      "oneWordSummary$": "ST_STM32F4_DISCOVERY",
      "description$": "ST_STM32F4_DISCOVERY",
      "settings": {
        "TOOLCHAIN_PREFIX": "C:/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update",
        "TARGET_SERIES": "STM32F4xx",
        "USE_FPU": "TRUE",
        "CHIBIOS_VERSION": "17.6.0",
        "CHIBIOS_SOURCE": "C:/ChibiOS",
        "CHIBIOS_BOARD": "ST_STM32F4_DISCOVERY",
        "NF_FEATURE_DEBUGGER": "TRUE",
        "API_Windows.Devices.Gpio": "ON",
        "API_Windows.Devices.Spi": "OFF",
        "NF_FEATURE_RTC": "ON"
      }
    },
    "ST_NUCLEO_F411RE": {
      "oneWordSummary$": "ST_NUCLEO_F411RE",
      "description$": "ST_NUCLEO_F411RE",
      "settings": {
        "TOOLCHAIN_PREFIX": "C:/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update",
        "TARGET_SERIES": "STM32F4xx",
        "USE_FPU": "TRUE",
        "CHIBIOS_VERSION": "17.6.0",
        "CHIBIOS_SOURCE": "C:/ChibiOS",
        "CHIBIOS_BOARD": "ST_NUCLEO_F411RE",
        "NF_FEATURE_DEBUGGER": "TRUE",
        "NF_FEATURE_RTC": "ON",
        "API_Windows.Devices.Gpio": "ON",
        "API_Windows.Devices.Spi": "OFF"
      }
    },
    "default$": "",
    "description$": "Variant for builds"
  }
}

networkfusion, 2017-09-11:
Let us start by updating that file first...

```
//cmake-variants.json
{
"buildType": {
"default$": "debug",
"description$": "the build type to use",
"debug": {
"oneWordSummary$": "Debug",
"description$": "Emit debug information without performing optimizations",
"buildType": "Debug"
},
"release": {
"oneWordSummary$": "Release",
"description$": "Enable optimizations, omit debug info",
"buildType": "Release"
},
"minsize": {
"oneWordSummary$": "MinSizeRel",
"description$": "Optimize for smallest binary size",
"buildType": "MinSizeRel"
},
"reldeb": {
"oneWordSummary$": "RelWithDebInfo",
"description$": "Perform optimizations AND include debugging information",
"buildType": "RelWithDebInfo"
}
},
"linkage": {
"ST_STM32F4_DISCOVERY": {
"oneWordSummary$": "ST_STM32F4_DISCOVERY",
"description$": "ST_STM32F4_DISCOVERY",
"settings": {
"TOOLCHAIN_PREFIX": "C:/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update",
"TARGET_SERIES": "STM32F4xx",
"USE_FPU": "TRUE",
"CHIBIOS_VERSION": "17.6.2",
"CHIBIOS_SOURCE": "",
"CHIBIOS_BOARD": "ST_STM32F4_DISCOVERY",
"NF_FEATURE_DEBUGGER" : "TRUE",
"API_Windows.Devices.Gpio=ON" : "ON",
"API_Windows.Devices.Spi" : "OFF",
"API_Windows.Devices.I2c" : "OFF",
"NF_FEATURE_RTC" : "ON",
"NF_FEATURE_USE_NETWORKING" : "OFF",
"SWO_OUTPUT" : "OFF"
}
},
"ST_NUCLEO_F411RE": {
"oneWordSummary$": "ST_NUCLEO_F411RE",
"description$": "ST_NUCLEO_F411RE",
"settings": {
"TOOLCHAIN_PREFIX": "C:/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update",
"TARGET_SERIES": "STM32F4xx",
"USE_FPU": "TRUE",
"CHIBIOS_VERSION": "17.6.2",
"CHIBIOS_SOURCE": "",
"CHIBIOS_BOARD": "ST_NUCLEO_F411RE",
"NF_FEATURE_DEBUGGER" : "TRUE",
"API_Windows.Devices.Gpio=ON" : "ON",
"API_Windows.Devices.Spi" : "OFF",
"API_Windows.Devices.I2c" : "OFF",
"NF_FEATURE_RTC" : "ON",
"NF_FEATURE_USE_NETWORKING" : "OFF",
"SWO_OUTPUT" : "OFF"
}
},
"default$": "",
"description$": "Variant for builds"
}
}
```

note that this version will automatically download the latest stable release (17.6.2) from the 17.6.x branch of chibios.

Also if that doesn't work, I am certain that changes have been made to the board definition files for most of the reference boards. It maybe necessary to update them to take this into account.

Lastly, this issue should really have been made in the "Home" repo and have followed the proper format... Hope this goes some way to resolving your issue though.

josesimoes, 2017-09-11:
@valoni a side comment: as the issue template describes new issues are to be open in the Home repo, not in the other ones. Please use the Home repo next time. Thanks! 😃

valoni, 2017-09-11:
where to get ChiBios 17.6.2 i can find just 17.6.0 to download manully it

i'm not download it by GIT tools ...

networkfusion, 2017-09-11:
there is no need to install git-tools, CMake should be able to download it. If not, you must delete the version (change "17.6.2" to "") and change back to your local copy that should have been Downloaded from the repo from https://github.com/nanoframework/ChibiOS/tree/stable_17.6.x

valoni, 2017-09-11:
**after update (latest source code . chibios) seems it not recognize CMAKE i can** 

Loading CMake Tools from c:\nf-interpreter\.vscode\.cmaketools.json
Detected available environment "Visual C++ 12.0 - x86
Detected available environment "Visual C++ 12.0 - amd64
Detected available environment "Visual C++ 12.0 - amd64_arm
Detected available environment "Visual C++ 14.0 - x86
Detected available environment "Visual C++ 14.0 - amd64
Detected available environment "Visual C++ 14.0 - amd64_arm
Started new CMake Server instance with PID 6920
Build program for generator Ninja is not found. Skipping...
Build program for generator Unix Makefiles is not found. Skipping...
None of preferred generators available on the system.

**when i tried to select SELECT BUILD SYSTEM showed me UNCONFIGURED PROJECT !!!**

valoni, 2017-09-11:
my mistake i forget to configure .vscode

networkfusion, 2017-09-11:
have you selected Visual C++ 14.0 - amd64_arm in the bottom right hand corner?

valoni, 2017-09-11:
yes it work well if i choose STM32F4Discovery but if i choose NUCLEO STM32F411RE i have this error from CHIBIOS 
-- CMake Error at CMakeLists.txt:495 (message):
 
  Sorry but couldn't find definition files for ST_NUCLEO_F411RE in the
  available list of ChibiOS supported boards...

  

-- Configuring incomplete, errors occurred!
See also "c:/nf-interpreter/build/CMakeFiles/CMakeOutput.log".
[vscode] Configure failed: [cmake-server] Configuration failed.

networkfusion, 2017-09-11:
that is likely because 

1. the boards files are not in step (lots of changes have occurred to the boards configurations recently for the new features that have been added)
2. the community-targets submodule is not up-to-date in the nf-interpreter repo
3. we have recently changed the order in which community-target boards are checked and I am not sure if this has been tested... @josesimoes @piwi1263 

Unfortunately keeping the community boards up-to-date is not the immediate concern of the core development team as keeping so many boards in sync at this early stage is time consuming and delays feature implementation, so this will have to be carried out by yourself (or another with interest in the board) so feel free to make a pull request on that front. We will try to make it easier by specifying which version of nf-interpreter it was last tested against...

josesimoes, 2017-09-11:
> we have recently changed the order in which community-target boards are checked and I am not sure if this has been tested

Yes, that was tested. 

Double check that `board.c` and `board.h` are located in the ST_NUCLEO_F411RE folder in the respective community target folder.

valoni, 2017-09-11:
ST_NUCLEO_F411RE  folder have inside board.c and board.h 
copied in the both folder in TARGETS and in TARGETS-COMMUNITY folder 
but nothing happen same error appear again

josesimoes, 2017-09-11:
This is now totally unrelated with the issue title!
Moving this to Slack [here](https://nanoframework.slack.com/archives/C4MGGBH1P/p1505147690000406)

piwi1263, 2017-09-12:
Hi guys, that board is renamed (and a lot of others) in chibios and must be named accordingly in the cmake-variants use ST_NUCLEO64_F411RE

networkfusion, 2017-09-15:
@valoni can this be closed?

valoni, 2017-09-15:
Yes please close it

valoni, 2017-09-15:
Yes please close it

closed 2017-09-15 by valoni